TousstNicolas/JLC2KiCad_lib #1: Mix between SMD and THT ends up in SMD only footprint

Bouni, 2021-12-30:
Hi,

first of all, this is absolutely amazing! I totally love this project 😍 

I tried to get footprint, symbol and 3D file for the [USB Micro connector]() `C132563 ` which should look like this:

[image]

The Symbol is correct:

[image]

But the footprint is not:

[image]

The 3D Package turned out great:

[image]

But there are two minor issues with the 3D package:

1. The path that is put into the footprint is an absolute path, so If I generate the part lets say in C:\temp\footprints\... , that path is put into the footprint. It breaks as soon as I move the package to another location.
2. The position of the 3D model in relation to the footprint is not correct and needs to be adjusted manually, but I have no idea if thats even automatically possible.

TousstNicolas, 2021-12-31:
Thank you for the comment and  reporting this issue. I corrected the missing holes for SMD-THT mixed components. 

I had no problem with the 3d model path on my side as long as the package is not moved, can you specify what the problem is on your side? 

I have added a `--model_path_relative` option which allows if it is present, to give the relative 3d model path. It is then formed as: `{output_dir/footprint_dir}/packages3d/{footprint_name}`
Let me now if you think there is a better solution.

Bouni, 2021-12-31:
Let me check this next week and I'll happily give you feedback! 
Keep up the good work 😊

Bouni, 2022-01-13:
Sorry for the late reply, I can confirm that the `--model_path_relative` switch works as expected!

closed 2022-01-13 by Bouni