TousstNicolas/JLC2KiCad_lib #2: .wrl file broken

Bouni, 2022-01-13:
I tried to download a part and its 3D model
`python JLC2KiCad_lib\JLC2KiCad_lib.py -dir jlcparts C475118 --model_path_relative` 

Which gave me this output:

```
2022-01-13 10:36:10,531 - INFO - creating library for component C475118
2022-01-13 10:36:10,889 - INFO - creating footprint ...
2022-01-13 10:36:11,501 - INFO - creating 3D model ...
2022-01-13 10:36:11,846 - INFO - added jlcparts/footprint/packages3d/KF141V-2.54-6P.wrl to footprint
2022-01-13 10:36:11,848 - INFO - created 'jlcparts/footprint/KF141V-2.54-6P.kicad_mod'
2022-01-13 10:36:12,200 - INFO - creating schematic KF141V-2.54-6P in default_lib
2022-01-13 10:36:12,201 - INFO - writing in jlcparts/Schematic/default_lib.lib file
```

If I use the .wrl file in KiCAD it is not shown and no error is given. If I open it in Windows using Print 3D I get an "VRML analysis error".

The part opened in an text editor is empty appart from some header comments:

```
#VRML V2.0 utf8
#created by JLC2KiCad_lib using the JLCPCB library
#for more info see https://github.com/TousstNicolas/JLC2KICAD_lib
```

Bouni, 2022-01-13:
After some searching I realized that easyeda is giving an error 500 for the 3D file:

`https://easyeda.com:443 "GET /analyzer/api/3dmodel/ef41c7d21d934cb3b1032c586dd1aa87 HTTP/1.1" 500 0`

TousstNicolas, 2022-01-13:
Yes, the requests to get it doesn't return anything.
I came to the same conclusion.

TousstNicolas, 2022-01-13:
I added an error message in case this appends for other components.

closed 2022-01-13 by Bouni