TousstNicolas/JLC2KiCad_lib #4: Some footprint/symbol name will break the file

andylinpersonal, 2022-01-19:
Symbol name containing whitespace characters and footprint name containing parentheses should be quoted by "".

TousstNicolas, 2022-01-19:
Thank you for reporting this issue. 
I corrected it, tell me if it works for you. 
If not, please give me the part number you are using so I can reproduce the issue.

andylinpersonal, 2022-01-25:
Oh, thank you. It is ok now. 

But there is another issue: these imported symbols are dismantled to several parts. 
Following screenshot is one of them LOL.
Part #: C2290, C72038

[image]

TousstNicolas, 2022-01-26:
Yes I know the schematic generation is far from ideal, but for those components I recommend you to use the schematic symbols from the standard library, and then link the imported footprint to them. 
I will work on this in the future when I have time. 
For now, I think the initial issue has been solved, so I am closing this issue. Do not hesitate to reopen it if you still encounter this issue.

closed 2022-01-26 by TousstNicolas