TousstNicolas/JLC2KiCad_lib #5: KiCad 6.0 symbol library format and .step 3D model export?

glowingkitty, 2022-01-27:
Thanks for the awesome work!:)
Any plans to add support for exporting symbols in the new library format? Otherwise one cannot edit the symbol in KiCad, except when having a separate library file for each part (as far as I know).
And being able to export 3D models as .step files would be super useful, for exporting 3D models from KiCad into other tools like Fusion360 for 3D modelling cases.

TousstNicolas, 2022-01-27:
Unfortunately for the 3D models, JLCPCB does not give us the .step files, but only an MTL file described as a raw unstructured triangulated surface, that I parse and convert to VRML for compatibility with KiCad. This means that unless JLCPCB change its file format (which is very unlikely), it will not be possible to export the 3D models as a .step files.

As for the symbol library, if you go to the symbol library management window, there is a "Migrate Libraries button" that you can use to migrate a Legacy library into the new .kicad_sym format supported by KiCad 6.0. Using this, you can generate the symbol and then migrate the library to the new format, you will then be able to correctly edit the symbol. See the screenshot below.

[image]

However, I will work on that in the future, the symbol creation is a mess and I will soon start a refactoring of this part, this will be the occasion to change the file format and export it in the new library format.

glowingkitty, 2022-01-27:
Regarding the symbol format: yeah, I was aware of that. But this also changes the file ending - and therefore JLC2KiCad_lib cannot add another part to an existing library as far as I know. So one has to create another library for every single part. It's a work around but not ideal.

And regarding .step files: to bad :( how about getting the .step files from [Octopart](https://octopart.com)? Of course it won't be possible for all parts - but for those parts which do exist on Octopart, it would be super useful to have both the 3D model from LCSC and Octopart.

TousstNicolas, 2022-02-03:
I opened a new branched for the kicad6.0 file format. I haven't finished every handler, but the most important and used ones are functional. 
Here is a list of known issues :
 - Pin number and text might be miss placed and/or have the wrong size and/or might not be hided. 
 - Text handler is not yet implemented.
 - Arc handler is not precise.
 - some other handlers are not implement as I haven't found any component using them. If you find any component with one of these handler, please report them so I can work on it. (you will get a warning message in this case)

I haven't got time to fully test it, so do not hesitate to report any other issues.

Overall the symbol generation should already be far more reliable than the current one. 
I will open another issue to keep track of issues and work to do.

To come back to  the 3D models, this is not in my priority list to get it from another website, the name used by JLCPCB and Octopart might not be the exact same and can lead to errors, plus it would add a lot of complexity to get it and I don't want the script to get to complicated to use.

glowingkitty, 2022-02-03:
Thanks for the update :)
Regarding 3D model: Yeah fair enough :)
Hm but could there maybe be a way to automate the process of this video here, to convert the 3D models into step files?

TousstNicolas, 2022-02-04:
I am not sure to understand which video you are talking about. But there are many tools online to convert .wrl file to .step or any other file type. Plus, even with this, you won't be able to edit the step file as it will still come from a wrl file which describe the triangular surface of the 3d model, which is fundamentally different than a .step file description.

What is your end objective with this conversion ? Is it to edit the 3d model, or only to preview it ? In the first case, it will not be possible to do it with fusion360 SOLIDWORKS nor freecad as they do not support this format. But you can do it with other software such as blender which is better suited for it.

glowingkitty, 2022-02-04:
Sorry, stupid me. Forgot the video link… https://youtube.com/watch?v=qSkWBwRJVqg

TousstNicolas, 2022-02-05:
I didn't realise that KiCad only exports .step models, I understand the problem better now. 
It's not one of my priorities, but I'll see when I have the time to implement a solution for this.

closed 2022-07-22 by TousstNicolas