TousstNicolas/JLC2KiCad_lib #9: Software broken: Symbol scaling suddenly wrong, import errors, footprints missing

glowingkitty, 2022-04-08:
I just installed the latest version via pip and while its awesome to see that the symbols are stored now in the new KiCad library format, I also noticed some urgent issues, making it currently impossible to use the plugin:

- Imported symbols are wayyy smaller then before, making it impossible to connect them to other components
[image]
- for some components I get an error like this one: 

> 2022-04-08 13:32:35,499 - INFO - created 'JLC2KiCad_lib/footprint/FS-1340.kicad_mod'
> 2022-04-08 13:32:35,846 - INFO - creating schematic FS-1340_0 in default_lib
> 2022-04-08 13:32:35,849 - INFO - creating library for component C2979183
> 2022-04-08 13:32:36,260 - INFO - creating footprint ...
> Traceback (most recent call last):
>   File "/usr/local/bin/JLC2KiCadLib", line 8, in <module>
>     sys.exit(main())
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/JLC2KiCadLib.py", line 128, in main
>     add_component(component, args)
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/JLC2KiCadLib.py", line 26, in add_component
>     footprint_name, datasheet_link = create_footprint(
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint.py", line 27, in create_footprint
>     footprint_name, datasheet_link, assembly_process = get_footprint_info(component_id)
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint.py", line 136, in get_footprint_info
>     response["data"]["componentPageInfo"]["list"][0]["componentModelEn"]
> IndexError: list index out of range

- some footprints (even including the one for the ESP32-WROOM-32E(4MB)) are missing, probably because of that error message?
- most footprints however seem correctly imported. With some exceptions, like this SMD nut, which also doesn't have 4 pins / connectors, but just one conducting body
[image]

TousstNicolas, 2022-04-09:
Thank you for reporting this issue.

I increased the default schematics size, i will later add an option to change it.
You can try it by updating the package using pip : 
pip install JLC2KiCadLib==1.0.1 
or 
pip install JLC2KiCadLib --upgrade

The footprint of ESP32-WROOM-32E(4MB) is, indeed broken, but it does not seem to be linked to the error you mentioned. I will work on this.
 
Do you have the reference of the components the error ?

TousstNicolas, 2022-04-09:
And I just found out for the ESP32-WROOM-32E(4MB) footprint, this issue came from the parenthesis in the footprint names, which where parsed by KiCad. I will replace these parentheses and the issue should not appear anymore.

glowingkitty, 2022-04-09:
awesome, thanks for the fast response and fixing it so quickly!:) Scaling is perfect again now.

Just re-downloaded the parts again. Some parts still get error messages. Any idea what's wrong here?
one example:

> 2022-04-09 14:35:39,381 - INFO - creating library for component C2879827
> 2022-04-09 14:35:39,787 - INFO - creating footprint ...
> 2022-04-09 14:35:40,628 - ERROR - footprint h_TRACK: layer correspondance not found
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", line 46, in h_TRACK
>     layer = layer_correspondance[data[1]]
> KeyError: '10'
> 2022-04-09 14:35:40,630 - ERROR - footprint h_TRACK: layer correspondance not found
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", line 46, in h_TRACK
>     layer = layer_correspondance[data[1]]
> KeyError: '10'
> 2022-04-09 14:35:40,630 - ERROR - footprint h_TRACK: layer correspondance not found
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", line 46, in h_TRACK
>     layer = layer_correspondance[data[1]]
> KeyError: '10'
> 2022-04-09 14:35:40,630 - ERROR - footprint h_TRACK: layer correspondance not found
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", line 46, in h_TRACK
>     layer = layer_correspondance[data[1]]
> KeyError: '10'
> 2022-04-09 14:35:40,630 - ERROR - footprint h_TRACK: layer correspondance not found
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", line 46, in h_TRACK
>     layer = layer_correspondance[data[1]]
> KeyError: '10'
> 2022-04-09 14:35:40,635 - INFO - creating 3D model ...
> 2022-04-09 14:35:41,001 - ERROR - request error, no 3D model found
> 2022-04-09 14:35:41,010 - INFO - created 'JLC2KiCad_lib/footprint/U261-121N-4BS2S.kicad_mod'

and another one

> 2022-04-09 14:36:04,385 - INFO - creating schematic SMTSO3015CTJ_0 in default_lib
> 2022-04-09 14:36:04,389 - INFO - creating library for component C2916320
> 2022-04-09 14:36:04,758 - INFO - creating footprint ...
> Traceback (most recent call last):
>   File "/usr/local/bin/JLC2KiCadLib", line 8, in <module>
>     sys.exit(main())
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/JLC2KiCadLib.py", line 128, in main
>     add_component(component, args)
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/JLC2KiCadLib.py", line 26, in add_component
>     footprint_name, datasheet_link = create_footprint(
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint.py", line 27, in create_footprint
>     footprint_name, datasheet_link, assembly_process = get_footprint_info(component_id)
>   File "/usr/local/lib/python3.9/site-packages/JLC2KiCadLib/footprint/footprint.py", line 136, in get_footprint_info
>     response["data"]["componentPageInfo"]["list"][0]["componentModelEn"]
> IndexError: list index out of range

that second one I get multiple times.

TousstNicolas, 2022-04-09:
The issue was present for components available on EasyEDA but not present on JLCPCB as I fetch some information on their website. I now fetch the datasheet link and the footprint from EasyEDA which solved the issue. 

I updated the package,  you can update it with  `pip install JLC2KiCadLib --upgrade`
I tested it on a few components and I didn't get any issues, let me know if you encounter some. 
If it works as expected, you can close this issue.

TousstNicolas, 2022-04-10:
After more testing, it seems that another error can occur. 

The footprints at esayEDA do not have exactly the same descriptions which can cause problems. 
I will work on it to solve them.

TousstNicolas, 2022-04-10:
Issue should be fixed, this component had a line on the 'Edge.Cuts' layer which I had not yet implemented. 

I updated the package, you can update it again with `pip install JLC2KiCadLib --upgrade`

If everything works properly, you can close this issue.

glowingkitty, 2022-04-11:
awesome!:) no python errors anymore.

Only those warnings/errors:

> 2022-04-11 16:01:46,476 - INFO - creating library for component C2979183
> 2022-04-11 16:01:46,969 - INFO - creating footprint ...
> 2022-04-11 16:01:47,416 - WARNING - footprint handler, h_TEXT: failed to add text
> 2022-04-11 16:01:47,416 - WARNING - footprint handler, h_TEXT: failed to add text
> 2022-04-11 16:01:47,418 - INFO - creating 3D model ...
> 2022-04-11 16:01:48,014 - ERROR - request error, no 3D model found
> 2022-04-11 16:01:48,021 - INFO - created 'JLC2KiCad_lib/footprint/BAT-TH_MY-18650-01-2.kicad_mod'
> 2022-04-11 16:01:48,436 - INFO - creating schematic MY-18650-01_0 in default_lib
> 2022-04-11 16:01:48,438 - INFO - creating library for component C2979182
> 2022-04-11 16:01:48,851 - ERROR - failed to get component uuid
> 2022-04-11 16:01:48,852 - INFO - creating library for component C383071

is there anything one can do about those errors? Or are they caused by an incomplete database from LCSC or JLCPCB?

TousstNicolas, 2022-04-11:
> also, KiCad gives me this error, when opening the footprint editor.
> 
> > Errors were encountered loading footprints:
> > Expecting 'at' in '/Users/..../Library/Mobile Documents/com~apple~CloudDocs/KiCad parts/JLC2KiCad_lib/footprint/ESP32-WROOM-32E(4MB).kicad_mod', line 7, offset 34.

Simply delete the file. I replaced the parenthesis in the footprint name (which were the cause of the errors), but your old broken file is this present in your library and KiCad is trying to load it.
The footprint should now be in the ESP32-WROOM-32E_4MB_.kicad_mod file.

> awesome!:) no python errors anymore.
>
> Only those warnings/errors:
>
>
> is there anything one can do about those errors? Or are they caused by an incomplete database from LCSC or JLCPCB?

- `ERROR - request error, no 3D model found` : mean that there is no 3D model available on their library. If you go on easyEDA, load the component and try to view the 3D model (view -> 3D view) you will not have the 3D model either. 
 
- `ERROR - failed to get component uuid` : Check the part number you are trying to import. There seems to be no component with the code C2979182  on easyEDA. You might have made a mistake. 

- `WARNING - footprint handler, h_TEXT: failed to add text`  : This is caused by an inconsistency in the request, I will try to handle it later. Meanwhile, you can manually add these two text label with the KiCad library editor.

glowingkitty, 2022-04-12:
ok, good to know :)

Since those IDs are all from LCSC, I tried to download all the parts which I bought from LCSC. Including [C2979182](https://lcsc.com/product-detail/Battery-Connectors_MYOUNG-MY-18650-02_C2979182.html). And yeah, that is one of the parts that doesn't have a footprint or a 3D model.
Just some fine tuning suggestion: it might make sense to replace the "ERROR - failed to get component uuid" message with something like "ERROR - no footprint found for C2979182", think that would be better understandable :)

closed 2022-04-12 by glowingkitty