TousstNicolas/JLC2KiCad_lib #12: Incorrect date in .kicad_sym file header?

bschwind, 2022-05-15:
Running KiCad version `(6.0.5-0), release build`

I'm using part number `C841192`

```bash
$ JLC2KiCadLib C841192
```

If I import the schematic library as-is, I get:

[image]

If I modify the date from `05052022` to `20211014`, it works fine.

If I modify the date to be `20220515` (as an example), KiCad complains that I need a newer version of KiCad in order to load the file.

The change was introduced [here](https://github.com/TousstNicolas/JLC2KiCad_lib/commit/4d5dce600f68ae99a417a9b762ee718da20ed905#diff-0173ed65391f3abba1bb7735bd7bf46fab12ac750f35797a4db5fcf4c89f0793R12) but I'm not entirely sure why.

TousstNicolas, 2022-06-01:
I have no idea either. 
I corrected it and uploaded it to PyPI, you can update it with `pip install JLC2KiCadLib==1.0.5` 
Thanks for reporting it, and I apologize for the  late response.  

I will let you close the issue if it works as expected now.

bschwind, 2022-06-02:
@TousstNicolas I ran `pip install JLC2KiCadLib==1.0.5` (and `pip install JLC2KiCadLib==1.0.5` just to be sure) and then `JLC2KiCadLib C841192` but I'm still seeing the incorrect date in the file.

I don't use python often so it's possible something with my environment is messed up.

> Thanks for reporting it, and I apologize for the late response.

No apologies needed! Thank you for this tool, it has already saved me a _ton_ of time, I appreciate it.

TousstNicolas, 2022-06-02:
Are you  using the same .kicad_sym file ? When you try to recreate/add a symbol, the script will append/replace the symbol description to the same file, and the header will not be recreated. Try  to delete the .kicad_sym library file and rerun the `JLC2KiCadLib C841192` command and let me know if this works.

bschwind, 2022-06-02:
@TousstNicolas I tried again after work today, with another invocation of `pip3 install JLC2KiCadLib==1.0.5`. It worked this time, so I'm not sure what was going on earlier in the day.

Thanks again for the fix, and for the tool!

closed 2022-06-02 by bschwind