TousstNicolas/JLC2KiCad_lib #14: Error when creating Symbol

elementzonline, 2022-06-17:
**Describe the bug**
 I am getting an error the error as described below
 
 ```
JLC2KiCadLib C85364 -schematic_lib ESD_C85364                                                                                                                     4389ms  Friday 17 June 2022 10:36:12 AM
2022-06-17 10:36:44,151 - INFO - creating library for component C85364
2022-06-17 10:36:45,340 - INFO - creating footprint ...
2022-06-17 10:36:46,364 - INFO - creating 3D model ...
2022-06-17 10:36:48,048 - INFO - added JLC2KiCad_lib/footprint/packages3d/SOT-23-6_L2.9-W1.6-P0.95-LS2.8-BR.wrl to footprint
2022-06-17 10:36:48,050 - INFO - created 'JLC2KiCad_lib/footprint/SOT-23-6_L2.9-W1.6-P0.95-LS2.8-BR.kicad_mod'
2022-06-17 10:36:49,101 - INFO - creating schematic SRV05-4-P-T7_0 in ESD_C85364
Traceback (most recent call last):
  File "/usr/local/bin/JLC2KiCadLib", line 8, in <module>
    sys.exit(main())
  File "/home/dhanish/.local/lib/python3.8/site-packages/JLC2KiCadLib/JLC2KiCadLib.py", line 129, in main
    add_component(component, args)
  File "/home/dhanish/.local/lib/python3.8/site-packages/JLC2KiCadLib/JLC2KiCadLib.py", line 38, in add_component
    create_schematic(
  File "/home/dhanish/.local/lib/python3.8/site-packages/JLC2KiCadLib/schematic/schematic.py", line 69, in create_schematic
    handlers.get(model)(args[1:], kicad_schematic)
  File "/home/dhanish/.local/lib/python3.8/site-packages/JLC2KiCadLib/schematic/schematic_handlers.py", line 289, in h_A
    cx, cy, theta, deltaTheta = getCenterParam(match)
  File "/home/dhanish/.local/lib/python3.8/site-packages/JLC2KiCadLib/schematic/schematic_handlers.py", line 254, in getCenterParam
    h = c(1, 0, n - e, a - t)
  File "/home/dhanish/.local/lib/python3.8/site-packages/JLC2KiCadLib/schematic/schematic_handlers.py", line 233, in c
    o = acos(i / r)
ZeroDivisionError: float division by zero
```

**To Reproduce**
Steps to reproduce the behavior:
1. LCSC part # that caused the issue - C85364
2. Arguments used for the execution - JLC2KiCadLib C85364 -schematic_lib ESD_C85364

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

TousstNicolas, 2022-06-17:
Hi, thank you  for reporting this issue.
This is an error I encountered in my automated tests, on which I applied the simple [Ostrich algorithm](https://en.wikipedia.org/wiki/Ostrich_algorithm). This is what happens when you parse data using simple split function instead of using the holy regex.
I will start writing some simple regex pattern to properly parse these requests' responses.

TousstNicolas, 2022-06-18:
Seems like the error was not coming from a bad parsing after all.
The error come from a [function](https://github.com/TousstNicolas/JLC2KiCad_lib/blob/7b6b285f46a0e6f9e40b7e66555c78e6c381a66a/JLC2KiCadLib/schematic/schematic_handlers.py#L227) reversed from the easyEDA javascript, in which I do not want to deep into to debug it. For now, I will patch this by catching the error so the symbol will be created without the arcs. 
You can update the package with `pip3 install JLC2KiCadLib==1.0.6` to get this last update.

elementzonline, 2022-06-19:
Wow 👍 Thanks for the quick action on this. It works well now. 

Closing the issue now ☺️

closed 2022-06-19 by elementzonline