TousstNicolas/JLC2KiCad_lib #18: Ability to use path variables for 3D models.

ste2425, 2022-09-26:
**Is your feature request related to a problem? Please describe.**
Working on my [PSP Consolizer](https://github.com/ste2425/PSP-Consolizer) project im trying to use path variables to keep my project portable. I'm using your great tool to allow me to use JLCPCB components however the paths to the 3D models appear to be absolute. This means to keep things portable i have to edit them to use variables.

**Describe the solution you'd like**
Some mechanism (too entirely sure how!) to allow path variables to be used for 3D models.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Manually editing paths after every component import.

Just wanted to add thanks for this brilliant tool, its been very helpful.

TousstNicolas, 2022-10-04:
Hi,

Thank you for submitting your feature request.

A possibility is to add an option to specify the base path of the 3D model with a variable.

Something like `JLC2KiCadLib C1337258 -model_base_variable MY_3DMODEL_LIBRARY`. The component library will still be created in the same place, but the 3D model will linked using `$(MY_3DMODEL_LIBRARY)/3Dfile.wrl`, so if you move the footprint later, the link to the 3D model will not be broken.

Would something like this works for you ?

ste2425, 2022-10-04:
That sounds like a very good idea.

Maybe fallback to the original behaviour if that argument isn't provided so it isn't a breaking change for other users.

TousstNicolas, 2022-10-04:
I implemented the solution mentioned before, you can try it by installing from source : 
```bash
git clone https://github.com/TousstNicolas/JLC2KiCad_lib.git
cd JLC2KiCad_lib 
pip install . 
```

I will update the PyPI repository after further testing.

closed 2022-10-26 by TousstNicolas