TousstNicolas/JLC2KiCad_lib #22: Component replace not working when pin_names is not created

Xyntexx, 2022-10-25:
Components in the .kicad_sym are not replaced but a duplicate is created each time the code runs.

### Steps to reproduce the behaviour:
Running code twice:
`JLC2KiCadLib.py C1337258 C24112 -dir My_lib -schematic_lib My_Schematic_lib --no_footprint`

### Why this happens
Running the code above creates the following line in `My_Schematic_lib.kicad_sym`:
`(symbol "STM32L431CCU6"   (in_bom yes) (on_board yes)`

The regex code on line in schematic.py, line 156:
`pattern = f'  \(symbol "{component_title}" \(pin_names (\n|.)*?\n  \)'`
does not recognize the line in the kicad_sym file because pin_names have not been created for some reason.

TousstNicolas, 2022-10-26:
I edited  the regex pattern, it should work properly now. 
I will let you close the issue if it works on your end.

Xyntexx, 2022-11-05:
It seems to be working. In what case is `(pin_names hide)` created?

closed 2022-11-05 by Xyntexx

TousstNicolas, 2022-11-05:
In EasyEDA, pins have a "Display Name" Field. The "(pin_names hide)" is present if this field is set to False. 
[image]