TousstNicolas/JLC2KiCad_lib #23: --model_path_relative argument not recognized

Xyntexx, 2022-10-25:
--model_path_relative argument not recognized even though it's in the usage in [README.md](https://github.com/TousstNicolas/JLC2KiCad_lib/blob/master/README.md).

Using the argument results in the following error:
```
usage: JLC2KiCadLib [-h] [-dir OUTPUT_DIR]
                    [-model_base_variable MODEL_BASE_VARIABLE]
                    [--no_footprint] [--no_schematic]
                    [-schematic_lib SCHEMATIC_LIB]
                    [-footprint_lib FOOTPRINT_LIB]
                    [-logging_level {DEBUG,INFO,WARNING,ERROR,CRITICAL}]
                    [--log_file]
                    JLCPCB_part_# [JLCPCB_part_# ...]
JLC2KiCadLib: error: unrecognized arguments: --model_path_relative
```

TousstNicolas, 2022-10-26:
Hi, 

Thank you for your many issues.

I did not update the README after problem #18. The --model_path_relative has been changed to [-model_base_variable MODEL_BASE_VARIABLE].

I will edit the README.

Xyntexx, 2022-11-05:
Good work!

closed 2022-11-05 by Xyntexx