TousstNicolas/JLC2KiCad_lib #24: Misaligned midpoint in some footprints

Xyntexx, 2022-11-06:
When using the preview of the JLCPCB assembly service some footprints are not positioned correctly. 
This issue appears on footprints that have a silkscreen that moves the midpoint of the footprint away from the midpoint of the pads.

Part#: **C2934560**

[image]
Original part.

[image]
After I moved the footprint midpoint to the centre of the pads.

[image]
The same problem appears on this USB-C connector, part# **C2765186**

I haven't tested this on any other than these two parts. I could not find a source on where the midpoint should be.

TousstNicolas, 2022-11-21:
Is it also the case with footprints not created by my script ? 
I found an issue that seems similar : https://github.com/Bouni/kicad-jlcpcb-tools/issues/143 
Looks like it might be an issue that come from the way JLC handles the kicad footprints, and does not come from how the  footprint is generated.

I will try to get deeper in the issue when I have more time.

Xyntexx, 2022-11-29:
I think I've created a working workaround where I search for the middle point of the pads. Should we create a branch for this?

TousstNicolas, 2022-11-29:
Yes, it might be better to create a new branch if we need to work on this. 
Thanks for the work  !

Xyntexx, 2023-01-16:
I've pushed the workaround to a branch in my fork. https://github.com/Xyntexx/JLC2KiCad_lib/tree/Pad__center

TousstNicolas, 2023-01-21:
So based on your changes, if I understand correctly, the issue was that JLCPCB only takes into account the pads when centering the component while I also took other parts  ? 

Are  there any cases where a footprints does not have any pads ? I don't recall ever encountering this case. It might be too much to keep track of the positions of the pads only, and every other parts in the footprint_info class. 

It also seems to bring issues with the 3D models centering, check with component C698918 for example where the X axis is not right (while correct with the current version).

Also, can you try do use black to format your code and keep it consistent with the rest.

As always, a big thanks for your work !

Xyntexx, 2023-01-21:
Is there any reason not to use the origin coordinates for the translation provided in the footprint data?
x and y from the data from C698918 
```
{
  "head": {
    "docType": "4",
    "editorVersion": "6.5.22",
    "c_para": {
      "package": "DO-218AB_L13.5-W8.4-LS15.5-FD",
      "pre": "D?",
      "Contributor": "lcsc",
      "link": "https://item.szlcsc.com/296833.html",
      "3DModel": "DO-218AB_L13.5-W8.4-LS15.5-FD"
    },
    "hasIdFlag": true,
    "x": 4032.4,
    "y": 3000.5,
    "utime": 1626858189,
    "uuid": "aa4c8bfeca62411b90acab80a9905e2f",
    "importFlag": 0,
    "transformList": ""
```

TousstNicolas, 2023-01-21:
It's  been a long time since I did this part, but from what I remember, These data where not matching the expected offset.

Xyntexx, 2023-01-22:
I've updated the https://github.com/Xyntexx/JLC2KiCad_lib/tree/Pad__center branch and used the origin coordinates provided in the data to keep the origin of the part where it's intended.
The main reason for me is to keep the footprint compatible with JLCPCB assemble service part origins. Apparently, EasyEDA automatically uses the middle of the 3d model and places it flat on the board regardless of the origin of the 3d model. Kicad uses the origin of the 3d model as the reference. With a few extra lines, we can calculate the midpoint of the 3d model and translate the 3d model accordingly. 
3D-model alignment should even improve when using this method.

It seems to work for me but let me know if you find some problems.

I added a black formatting GitHub action to my main branch just to test it. That might be a good way to make sure the contributed code is on par with the repository requirements.  It can even make a commit or a pull request with the correctly formatted code. 😉

TousstNicolas, 2023-01-22:
That's a lot of stuff done, thank you ! 

I will try to test everything today.

TousstNicolas, 2023-01-24:
merged in 12c6860aecd43dd8b4262347f74a3bb06de85959

closed 2023-01-24 by TousstNicolas