TousstNicolas/JLC2KiCad_lib #27: Adding a symbol to the library is sometimes added outside the structure

Xyntexx, 2022-12-10:
Something in my workflow sometimes adds a new line at the end of the symbol library file. 
It might be git or modifying the library in KiCad.
Because of the following line, the new symbol will be added outside the library structure and will not be visible in KiCad. 
https://github.com/TousstNicolas/JLC2KiCad_lib/blob/3920e1ff374599895b2034e14e00f5258b73f968/JLC2KiCadLib/schematic/schematic.py#L181

Xyntexx, 2022-12-10:
Please test

TousstNicolas, 2022-12-10:
There were a small offset issue that caused the last closing parenthesis to be taken off. I also used the footer instead of the single ")", but it is essentially the same. 
As always, thanks for your contribution !

closed 2022-12-11 by Xyntexx