TousstNicolas/JLC2KiCad_lib #29: division by zero at line 218 in footprint_handlers.py

cmteuffel, 2022-12-20:
Dear Nicolas,

first of all thanks for sharing your tool with the world. It is really useful and well done. I also know that you hoped to not touch it again, but I found an error which seems to occur mainly with SMD capacitors. The error seems to happen in the footprint generation but the kicad_sym file gets also corrupted with the respective symbol tag not being closed. I hope you find the time to take a look into it.

Best regards,
Christian

Here is the error output

> 2022-12-19 16:18:34,584 - INFO - created 'My_lib/footprint/CAP-SMD_L7.3-W4.3-R-RD.kicad_mod'
> 2022-12-19 16:18:35,080 - INFO - creating schematic TAJE107M025RNJ_0 in My_Schematic_lib
> 2022-12-19 16:18:35,081 - INFO - creating library for component **C312983**
> 2022-12-19 16:18:35,576 - INFO - creating footprint ...
/home/lithium/.local/lib/python3.10/site-packages/KicadModTree/util/geometric_util.py:327: UserWarning: Start and end point are not an same arc.
>                     Extended line from center to end point used to determine angle.
>   warnings.warn(
> 2022-12-19 16:18:36,072 - ERROR - footprint handler, h_ARC: failed to add ARC
> Traceback (most recent call last):
>   File ".local/lib/python3.10/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", **line 218, in h_ARC**
    centerX = ((start[1] - end[1]) / (start[1] - midpoint[1]) * sq1 - sq2) / (
> ZeroDivisionError: **float division by zero**
> 2022-12-19 16:18:36,073 - ERROR - footprint handler, h_ARC: failed to add ARC
> Traceback (most recent call last):
>   File ".local/lib/python3.10/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", **line 218, in h_ARC**
    centerX = ((start[1] - end[1]) / (start[1] - midpoint[1]) * sq1 - sq2) / (
> ZeroDivisionError: **float division by zero**
> 2022-12-19 16:18:36,073 - INFO - creating 3D model ...
> 2022-12-19 16:18:36,878 - INFO - added [..]JLC2KiCad_lib/My_lib/footprint/packages3d/C1206.wrl to footprint
> 2022-12-19 16:18:36,880 - INFO - created 'My_lib/footprint/C1206.kicad_mod'
> 2022-12-19 16:18:37,374 - INFO - creating schematic GRM31CR61A107ME05L_0 in My_Schematic_lib
> 2022-12-19 16:18:37,375 - INFO - creating library for component **C96123**
> 2022-12-19 16:18:37,871 - INFO - creating footprint ...
> 2022-12-19 16:18:38,360 - ERROR - footprint handler, h_ARC: failed to add ARC
> Traceback (most recent call last):
>   File ".local/lib/python3.10/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", **line 218, in h_ARC**
    centerX = ((start[1] - end[1]) / (start[1] - midpoint[1]) * sq1 - sq2) / (
> ZeroDivisionError: **float division by zero**
> 2022-12-19 16:18:38,360 - ERROR - footprint handler, h_ARC: failed to add ARC
> Traceback (most recent call last):
>   File ".local/lib/python3.10/site-packages/JLC2KiCadLib/footprint/footprint_handlers.py", **line 218, in h_ARC**
    centerX = ((start[1] - end[1]) / (start[1] - midpoint[1]) * sq1 - sq2) / (
> ZeroDivisionError: **float division by zero**

TousstNicolas, 2022-12-20:
Hi,
Thank you for reporting this issue.
I don't have a lot of time for now, but I will try to look into it when I can.

cmteuffel, 2022-12-21:
C129131 is another component where the conversion fails with the same error.

cmteuffel, 2022-12-21:
I just found out that the corruption of the kicad_sym file is caused by a different bug and by different parts in my partlist -> #30

TousstNicolas, 2023-01-24:
fixed with 12c6860aecd43dd8b4262347f74a3bb06de85959

closed 2023-01-24 by TousstNicolas