TousstNicolas/JLC2KiCad_lib #33: Proper Arc support for schematic and footprint

TousstNicolas, 2023-01-22:
The arcs for the schematic and footprint are rarely properly placed. It needs to be correctly handled.

I pushed some work on the Fix_ARC branch, but I cannot figure out how to draw them, there are always components for which it does not work.

The following components have an arc in their footprint and could be used to test:
C55684 C185659 C86002 C312983 C1341701 C307522 C689358 C403695 C602208 C152951 C688068 C163798 C661330

For now, I tried to use the GetCenterParam function, which is reversed from https://easyeda.com/editor/6.5.5/js/editorPCB.min.js
This function seems to return the coordinates of the center (or possibly the midpoint in some occasions ? ), and two angles, but when using theses to calculate the start point, end point and center, the result is inconsistent, sometimes it works, sometimes it doesn't.

Xyntexx, 2023-01-22:
I made my version of the code and it seems to be working pretty well so far:
https://github.com/Xyntexx/JLC2KiCad_lib/tree/my_ARC_Fixes

TousstNicolas, 2023-01-24:
Thanks a lot for your work, merged in 12c6860aecd43dd8b4262347f74a3bb06de85959

closed 2023-01-24 by TousstNicolas