TousstNicolas/JLC2KiCad_lib #34: Poor symbol origin placement

Xyntexx, 2023-01-22:
For some parts the symbol is very far from the origin:

C82942
[image]

C6649
[image]

C698918
[image]

Any ideas what's causing this?

TousstNicolas, 2023-01-22:
Yes it is something that happens on a lot of components. The eeschema center it automatically once placed so I didn't pushed to find the issue here. 

EasyEDA use a different coordinate system, so for every schematic, I need to scale and translate everything. 

for example with the pins : 
https://github.com/TousstNicolas/JLC2KiCad_lib/blob/bb11e2d853b4c3ae273d344b61636c0bbf725295/JLC2KiCadLib/schematic/schematic_handlers.py#L95-L96

For most components it works fine, but for others, it is completely off-centerer. 
There might be a flag or an  additional translation indications somewhere in the data, but I did not found it.

TousstNicolas, 2023-01-22:
I think I found the missing data, I will test if this works consistently and patch it.

closed 2023-01-23 by TousstNicolas