TousstNicolas/JLC2KiCad_lib #35: Use library to convert json files to footprints/symbols

Xyntexx, 2023-01-24:
It would be nice to be able to use the library for converting json files exported from easyEDA to create footprints and symbols for kicad.

This would also help developing the code since its possible to create custom footprints for testing.

TousstNicolas, 2023-01-24:
I am not sure to understand what you mean, could you give me more details please ?

Xyntexx, 2023-01-24:
Instead of creating Kicad libraries using the JLCPCB part number the same code could be used to convert footprints exported from the EasyEDA editor online. The export option produces a JSON file that uses the same format as the JSON fetched from the library. 

Here is how to export a footprint from EasyEDA
[image]

Here is the JSON code from the footprint above
```
{
  "head": {
    "docType": "4",
    "editorVersion": "6.5.22",
    "newgId": true,
    "c_para": {
      "package": "",
      "pre": "U?",
      "Contributor": "Guest"
    },
    "hasIdFlag": true,
    "x": 4000,
    "y": 3000
  },
  "canvas": "CA~1000~1000~#000000~yes~#FFFFFF~7.874~1000~1000~line~3.937~mm~1~45~~3.937~4000~3000~0~none",
  "shape": [
    "TRACK~1~2~~4055.118 2976.378 4055.118 2984.252~gge244~0",
    "RECT~4000~3000~39.37~39.37~3~gge10~0~1~none~~~",
    "RECT~4047.244~3000~39.37~39.37~3~gge16~0~0~~~~",
    "SOLIDREGION~3~~M 4070.866 2984.252 L 4078.74 2976.378 L 4078.74 2968.504 L 4070.866 2976.378 L 4070.866 2984.252 Z~solid~gge247~~~~0",
    "TRACK~1~1~~4055.118 2992.126 4055.118 2984.252~gge223~0",
    "DIMENSION~12~M4086.614,3027.559L4066.929,3027.559M4080.614,3026.059L4086.614,3027.559L4080.614,3029.059M4072.929,3026.059L4066.929,3027.559L4072.929,3029.059M 4050.04 3010.97L4045.5 3010.97 L4045.04 3015.06 L4045.5 3014.6 L4046.86 3014.15 L4048.23 3014.15 L4049.59 3014.6 L4050.5 3015.51 L4050.95 3016.88 L4050.95 3017.79 L4050.5 3019.15 L4049.59 3020.06 L4048.23 3020.51 L4046.86 3020.51 L4045.5 3020.06 L4045.04 3019.6 L4044.59 3018.7 M 4054.41 3018.24L4053.95 3018.7 L4054.41 3019.15 L4054.86 3018.7 L4054.41 3018.24 M 4060.59 3010.97L4059.23 3011.42 L4058.32 3012.79 L4057.86 3015.06 L4057.86 3016.42 L4058.32 3018.7 L4059.23 3020.06 L4060.59 3020.51 L4061.5 3020.51 L4062.86 3020.06 L4063.77 3018.7 L4064.23 3016.42 L4064.23 3015.06 L4063.77 3012.79 L4062.86 3011.42 L4061.5 3010.97 L4060.59 3010.97 M 4069.95 3010.97L4068.59 3011.42 L4067.68 3012.79 L4067.23 3015.06 L4067.23 3016.42 L4067.68 3018.7 L4068.59 3020.06 L4069.95 3020.51 L4070.86 3020.51 L4072.23 3020.06 L4073.14 3018.7 L4073.59 3016.42 L4073.59 3015.06 L4073.14 3012.79 L4072.23 3011.42 L4070.86 3010.97 L4069.95 3010.97 M 4079.32 3010.97L4077.95 3011.42 L4077.04 3012.79 L4076.59 3015.06 L4076.59 3016.42 L4077.04 3018.7 L4077.95 3020.06 L4079.32 3020.51 L4080.23 3020.51 L4081.59 3020.06 L4082.5 3018.7 L4082.95 3016.42 L4082.95 3015.06 L4082.5 3012.79 L4081.59 3011.42 L4080.23 3010.97 L4079.32 3010.97 M 4085.95 3014.15L4085.95 3020.51 M 4085.95 3015.97L4087.32 3014.6 L4088.23 3014.15 L4089.59 3014.15 L4090.5 3014.6 L4090.95 3015.97 L4090.95 3020.51 M 4090.95 3015.97L4092.32 3014.6 L4093.23 3014.15 L4094.59 3014.15 L4095.5 3014.6 L4095.95 3015.97 L4095.95 3020.51 M 4098.95 3014.15L4098.95 3020.51 M 4098.95 3015.97L4100.32 3014.6 L4101.23 3014.15 L4102.59 3014.15 L4103.5 3014.6 L4103.95 3015.97 L4103.95 3020.51 M 4103.95 3015.97L4105.32 3014.6 L4106.23 3014.15 L4107.59 3014.15 L4108.5 3014.6 L4108.95 3015.97 L4108.95 3020.51 M 4086.614 3039.37 4086.614 3017.559 M 4066.929 3039.37 4066.929 3017.559~gge301~~0~level~0.4",
    "ARC~1~5~~M4023.622,2960.63 A7.874,7.874 0 0 0 4007.874,2960.63~~gge256~0",
    "ARC~1~5~~M4023.622,2952.756 A12.8829,12.8829 0 0 0 4007.874,2948.819~~gge268~0",
    "ARC~1~5~~M4027.5588,2959.4769 A9.1821,9.1821 0 0 0 4033.1265,2946.0352~~gge265~0",
    "ARC~1~5~~M4039.37,2944.882 A7.874,7.874 0 1 1 4039.37,2960.63~~gge259~0",
    "PAD~RECT~4015.748~2992.126~6~6~11~~1~1.9685~4012.748 2989.126 4018.748 2989.126 4018.748 2995.126 4012.748 2995.126~0~gge28~0~~Y~0~~~4015.748,2992.126",
    "PAD~ELLIPSE~4000~2992.126~6~6~11~~2~1.9685~~0~gge46~0~~Y~0~~~4000,2992.126",
    "PAD~OVAL~4031.496~2992.126~6~11.811~11~~3~1.9685~4028.9798 2990.6733 4034.0122 2993.5788~60~gge64~0~~Y~0~~~4031.496,2992.126",
    "PAD~POLYGON~4047.244~2992.126~7.874~6.819~11~~4~1.9685~4045.2755 2988.7165 4049.2125 2988.7165 4051.181 2992.126 4049.2125 2995.5355 4045.2755 2995.5355 4043.307 2992.126~0~gge82~0~~Y~0~~~4047.244,2992.126",
    "PAD~POLYGON~4047.244~2976.378~7.874~6.819~11~~4~0.9843~4045.2755 2972.9685 4049.2125 2972.9685 4051.181 2976.378 4049.2125 2979.7875 4045.2755 2979.7875 4043.307 2976.378~0~gge163~3.937~4046.2598 2976.378 4048.2282 2976.378~Y~0~~~",
    "PAD~OVAL~4031.496~2976.378~6~6~11~~3~0.9843~4031.496 2976.378 4031.496 2976.378~0~gge166~3.937~4030.5118 2976.378 4032.4802 2976.378~Y~0~~~",
    "PAD~ELLIPSE~4000~2976.378~6~6~11~~2~0.9843~~0~gge169~3.937~3999.0158 2976.378 4000.9842 2976.378~Y~0~~~",
    "PAD~RECT~4015.748~2976.378~6~6~11~~1~0.9843~4011.5054 2976.378 4015.748 2972.1354 4019.9906 2976.378 4015.748 2980.6206~45~gge172~3.937~4015.0521 2977.0739 4016.4439 2975.6821~Y~0~~~",
    "VIA~4055.118~2984.252~2.4~~0.6~gge184~0",
    "VIA~4055.118~2976.378~2.4~~0.6~gge196~0",
    "VIA~4055.118~2992.126~2.4~~0.6~gge208~0",
    "HOLE~4062.992~2964.567~4~gge283~0"
  ],
  "layers": [
    "1~TopLayer~#FF0000~true~false~true~",
    "2~BottomLayer~#0000FF~true~false~true~",
    "3~TopSilkLayer~#FFCC00~true~false~true~",
    "4~BottomSilkLayer~#66CC33~true~false~true~",
    "5~TopPasteMaskLayer~#808080~true~true~true~",
    "6~BottomPasteMaskLayer~#800000~true~false~true~",
    "7~TopSolderMaskLayer~#800080~true~false~true~0.3",
    "8~BottomSolderMaskLayer~#AA00FF~true~false~true~0.3",
    "9~Ratlines~#6464FF~false~false~true~",
    "10~BoardOutLine~#FF00FF~true~false~true~",
    "11~Multi-Layer~#C0C0C0~true~false~true~",
    "12~Document~#FFFFFF~true~false~true~",
[... cut ...]
    "Copper_Area~true~true",
    "Circle~true~true",
    "Arc~true~true",
    "Solid_Region~true~true",
    "Text~true~true",
    "Image~true~true",
    "Rect~true~true",
    "Dimension~true~true",
    "Protractor~true~true"
  ],
  "BBox": {
    "x": 3996.8,
    "y": 2944.9,
    "width": 112.2,
    "height": 94.5
  },
  "netColors": {}
}
```

I haven't needed to import custom or user-contributed footprints from EasyEDA other than for potentially debugging for this library but it could be achieved with low effort.

TousstNicolas, 2023-01-25:
I understand the usefulness of this feature for debugging. 

However, I don't see a practical use case for it and implementing it would add unnecessary complexity to the script. As you mentioned, you have not personally needed it in the past.

I will be closing this issue as it is out of the scope of the current project. If anyone else feels that they have a compelling use case for this feature and finds this issue, they are welcome to reopen it and provide their reasoning.

Thank you

closed 2023-01-25 by TousstNicolas