TousstNicolas/JLC2KiCad_lib #38: ideas

immortal-sniper1, 2023-02-26:
**Additional context**
Add any other context or screenshots about the feature request here.
Is it possible to import from JLC also the assembly type tag? aka SMD or wave soldering ?
Also if it is a Basic part or not? this can really help a lot.

Also but this is more like a side topic i noticed that every part comes with its own footprint , even tho that footprint is a standard one this may make the lib grow in size very fast due to a lot of duplicates.

BTW i also found it a bit unclear how the parameters in the command line work so maybe a bit more explanations there would be nice

TousstNicolas, 2023-02-26:
> Is it possible to import from JLC also the assembly type tag? aka SMD or wave soldering ?
Also if it is a Basic part or not? this can really help a lot.

This shoud be possible, I will try to  take a look at it when I have the time. 

> Also but this is more like a side topic i noticed that every part comes with its own footprint , even tho that footprint is a standard one this may make the lib grow in size very fast due to a lot of duplicates.

You can use the ` --no_footprint` argument if you do not want to create the footprint. 
 for example : `JLC2KiCadLib C1337258 --no-footprint` will only create the schematic for the mentioned component. 
 
 > BTW i also found it a bit unclear how the parameters in the command line work so maybe a bit more explanations there would be nice
 
 I will try  to update the readme when I have the time to give better explanations.

immortal-sniper1, 2023-02-26:
schematic == symbol for kicad here is another source of confusion since i understand schematic as being the entire circuit with op amps resistors and such 

btw regarding footprint can u get a list in a script? since i think you can use string comparison (and that may be enough for most parts aka resistors and caps but i think it may work for many other standard footprintss)

mikaelj, 2023-05-09:
@immortal-sniper1 meanwhile, you can use this simple bash script I wrote to do batch processing of multiple parts, to add the missing SMD attr: https://github.com/mikaelj/jlc-kicad/blob/19302e96aa4621850ff5f9dfa071db96adff1f28/jlcpcb-library.sh#L44

closed 2023-05-14 by TousstNicolas