TousstNicolas/JLC2KiCad_lib #41: PCB cutout not included

ReinBentdal, 2023-03-19:
**Describe the bug**
PCB cutout not included when creating footprint

**To Reproduce**
Steps to reproduce the behavior:
1. Load part C559118 in EasyEDA, use 3D viewer to see a cutout in the part
2. Create footprint with this lib, and load into KiCad. There is no hole

**Expected behavior**
I would expect the lib to also be able to include cutouts in the parts

**Screenshots**
EasyEDA:
[image]

KiCad:
[image]

Trajelectory, 2023-03-23:
i'm not sure it's an issue with this script but with Kicad himself.

TousstNicolas, 2023-03-25:
Hi, thanks for reporting this.

It is indeed missing as EasyEDA consider the cutout as not just a regular shape on the Edge.Cuts layer, but as a solidregion which was not handled. 

I added a handler for the simple polygons cutouts, unfortunately I do not have the time to look deeper into it and try to understand the logic of how EasyEDA parse the more complex shapes.  

The polygons function correctly once they are placed on a board, but the 3D preview may not display correctly in some instances. I am unsure of the reason for this issue :

footprint 3D preview : 
[image]

PCBnew 3D viewer :
[image]

Help would be greatly appreciated to handle more complex shapes and solve the previous issue.
The component  `C2687297`  contains shapes that can be used to work on this. 

If you are want to help, the current handler can be found in the footprint handler file :
https://github.com/TousstNicolas/JLC2KiCad_lib/blob/fd492d0f21c526d10292b3d26dc50af3f3c9b13f/JLC2KiCadLib/footprint/footprint_handlers.py#L329-L337

ReinBentdal, 2023-03-27:
Thanks for you effort! I will check out the new version when I have time. Maybe if I have time in the future, and feel the need, I will investigate

TousstNicolas, 2026-01-04:
Fully implemented with [4f14e2c](https://github.com/TousstNicolas/JLC2KiCad_lib/commit/4f14e2c1cea6a3c7695356ff9abd022e1ca1fbfc)

closed 2026-01-04 by TousstNicolas