TousstNicolas/JLC2KiCad_lib #46: Fix for #27 (appending item outside of structure) was lost

vvuk, 2023-05-23:
It looks like the fix that was merged in issue #27 was lost; it's not present in the current symbol.py (and the problem occurs).

vvuk, 2023-05-23:
Digging more into this -- it looks like even if I bring back the fix, it's not correct; the issue now seems to be that my existing library file is `43191` bytes, but when the contents are read into python, the length of the string is `43188`. This is presumably because there are some actual UTF-8 sequences in there... so a `rfind` will find the right string index for the paren, but will be the wrong offset for the `lseek()`. 

I've locally changed this to just do the append as a string and write out the file:

```
            new_content = file_content[:file_content.rfind(")")]
             new_content = new_content + template_lib_component + template_lib_footer
             lib_file.seek(0)
             lib_file.write(new_content.encode())
```

not great, but it solves the issue.

TousstNicolas, 2023-08-20:
Thanks for reporting this  issue.
This is indeed an issue I already encountered before. 
I did not found any better solution. So I will do it as you did. It should solve another similar issue.

closed 2023-08-20 by TousstNicolas