TousstNicolas/JLC2KiCad_lib #48: After editing symbols library, parts can't be added.

jonsag, 2023-08-12:
**Describe the bug**
After importing one or more parts and editing or removing parts, parts added doesn't show.

**To Reproduce**
Steps to reproduce the behavior:
1. LCSC part # that caused the issue
3. Arguments used for the execution

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

TousstNicolas, 2023-08-20:
Hi, 
Thank you  for reporting this issue. 
It should be fixed with the latest commit. 
I will let you close the issue if you can confirm this is working as expected on your side.

jonsag, 2023-08-20:
Great.
Haven't got the possibility to test at the moment, but as soon as
everything is OK, I will close it.

Thanks!
/jon
------------------------------------------------------------------------------
Jon Sagebrand

Jonix
Blacksta 6
611 95 Nyköping

***@***.***

070-694 67 34
------------------------------------------------------------------------------

jonsag, 2023-08-22:
Seems to be working as it should now.
Good work!
Thanks

closed 2023-08-22 by jonsag