TousstNicolas/JLC2KiCad_lib #57: Cannot connect symbol pins

dzid26, 2023-11-20:
**Describe the bug**
Wire doesn't connect to symbol pins

**To Reproduce**
Steps to reproduce the behavior:
1. LCSC part # that caused the issue C2919500 
 

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Additional context**
I tried "aligning the schematic and also the symbol drawing in the editor. Didn't help.  
[image]

dzid26, 2023-11-20:
Oh, after moving the pin a bit 
[image]
and then back "where it was", the problem was fixed. 

I have a hunch that some iterative error accumulation is happening when pins are populated and only far top and far bottom pins are affected.

TousstNicolas, 2023-11-21:
I rounded the position and length of the pins to the third digit, it should solve those imprecision errors.  

It works on C2919500.

I updated the Pip package, you  can test the update with `pip3 install JLC2KiCadLib==1.0.28`

I will let you close the issue if it works on your side.

closed 2023-11-22 by dzid26