TousstNicolas/JLC2KiCad_lib #58: Kicad plugin

dzid26, 2023-12-01:
**Is your feature request related to a problem? Please describe.**
Command line based footprint based downloading and placing on pcb is a little cumbersome

**Describe the solution you'd like**
GUI kicad plugin shall support:
- part selection and download
  - temporary footprint fitting on the PCB
  - saving part to a project library
- removing not needed parts from library

**Additional context**
I hacked up together a plugin that downloads part to a hardcoded library name in the project directory, and [pastes](https://forum.kicad.info/t/python-place-move-footprint-with-a-mouse/46332) it on PCB canvas.
Currently it can be [installed](https://github.com/Springer-Electronics/JLC2KiCad_lib#install-as-kicad-script-plugin) directly in kicad scripts directory. 

https://github.com/TousstNicolas/JLC2KiCad_lib/assets/841061/cd7311e8-9809-43a6-ab2e-ba5304feb2f0

dzid26, 2023-12-10:
I [added](https://github.com/Springer-Electronics/JLC2KiCad_lib/commits/kicad-plugin/) possibility to quickly paste the footprint or save to the library:
[image]

TousstNicolas, 2023-12-12:
Hi,

This is something I considered a few years back.
At that time, I created a complete standalone GUI in which you could browse the JLCPCB library and configure everything needed.
Then a [KiCad issue](https://gitlab.com/kicad/code/kicad/-/issues/2077) was created regarding a possible Eeschema API. So I postponed it and planned to release it as a fully functional plugin when it became available.

Unfortunately, I don't have enough time today to work on it.

I may try to revive it and release it "as is" with a few updates needed if I still find the time.

markusdd, 2024-03-10:
FYI: I made a UI now which makes the process more convenient.

Maybe you guys want to check it? I stepped away from doing it in wxwidgets directly in KiCAD because there is not much gain having it in there, but maintaining it is kind of a nightmare.

https://github.com/markusdd/EasyEDA_to_KiCAD_Lib_UI

I released a first windows build, more to follow.