TousstNicolas/JLC2KiCad_lib #62: Micro-USB Footprint: THT elements are missing in the footprint

besi, 2024-02-10:
**Describe the bug**
I'm using a Micro USB Plug `JLC#C505111`. I do have a working footprint from the Kicad Standard Library for comparison. 
The Kicad Footprint is `Connector_USB:USB_Micro-B_Molex-105017-0001`

Now the EasyEDA-Converted footprint is missing a slot which goes through the PCB.
EasyEDA Preview shows the correct footprint:
https://jlcpcb.com/partdetail/Molex-1050171001/C505111

[image]

However the downloaded footprint is missing the mentioned slot (Pad 8 on the EasyEDA preview):
[image]
[image]

Another issue is that the round holes in the back (Pin6 and Pin7) also do not show up on the 3d Viewer so it's uncertain if those will get manufactured too.

I just started using this tool and I am very pleased you made the effort to create such a useful program. 
So thanks very much and thanks for looking into this issue ;-)

TousstNicolas, 2024-02-11:
Hi, thanks for reporting this issue.

From the logs I see a beautiful `WARNING - footprint handler, pad : unknown assembly_process skiping pad n° : 8`. (with a little typo in it :open_mouth:  ) 

I will start looking into it.

TousstNicolas, 2024-02-13:
The fix is available with version 1.0.30.
You should be able to update to this version.

I will let you close this issue if it works on your side.

[image]

closed 2024-02-16 by TousstNicolas