TousstNicolas/JLC2KiCad_lib #63: Incorrect pin assignment when importing component C12087

carlsa, 2024-03-01:
**Describe the bug**
The function of pin 1 and 3 are swapped when I import component C12087.

In EasyEDA, the pin assignment is:
Pin 1 - `GND`
Pin 3 - `In`

In the KiCAD library those two pins are swapped around
```
(pin unspecified line
  (at -12.7 1.27 0)
  (length 5.08)
  (name "In" (effects (font (size 1 1))))
  (number "1" (effects (font (size 1 1))))
)
(pin unspecified line
  (at 0.0 -6.35 90)
  (length 5.08001016002032)
  (name "GND" (effects (font (size 1 1))))
  (number "3" (effects (font (size 1 1))))
)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Import component C12087 with the command below
`JLC2KiCadLib C12087 --skip_existing -dir lib -symbol_lib test2 -footprint_lib test2`
2. This is my output: [Output_from_command_C12087.zip](https://github.com/TousstNicolas/JLC2KiCad_lib/files/14458283/Output_from_command_C12087.zip)

**Expected behavior**
That the pins aren't swapped.

**Screenshots**
[image]
[image]

**Additional context**
I have verified that the imported pin assignment would result in an incorrect assignment to the footprint (i.e. that the footprint pads aren't also swapped)

I have tested JLC2KiCadLib 1.0.21 where I saw the same behavior.

`python --version`
Python 3.12.2

I'm running from this docker file; if you think it is useful you are very welcome to add it to the project!
https://github.com/carlsa/JLC2KiCad_docker

TousstNicolas, 2024-03-02:
Hi Thanks for reporting this issue ! 

In EasyEDA, when looking at the components pin details, we can see the following : 
[image]

It seems like the pin number and spice pin number are inverted for the IN and GND pins.
Therefor the displayed pin number in easyEDA might be the incorrect pin number, and the ones in kicad the correct ones. 

For JLC2KiCad_Lib I use the spice number as the pin number the create the schematic, which explains why the displayed number is the correct one. 

Let me know what you think.

carlsa, 2024-03-02:
The result on the PCB appears to be incorrect when I compare with the [data sheet](http://www.advanced-monolithic.com/pdf/ds1117.pdf).

[image]

Is the pin number also available from the API? I'm thinking a solution could be to throw a warning or include a warning text in the symbol if there's a discrepancy.

TousstNicolas, 2024-03-19:
The pin number is often different from the pin names on easyEDA, so triggering a warning to often would render it useless.

I still consider it as an issue on easyEDA's side and since it appears to  be very rare, I will not implement anything to handle this behavior.