TousstNicolas/JLC2KiCad_lib #65: Symbol labels missing when importing part C150858

markusdd, 2024-03-05:
Part C150858 (Skyworks RFX2411) in a QFN-20 package is not being generated properly.

The designations seem correct, but all the visual pin labels are missing:
[image]

The EasyEDA symbol has them:
[image]

I have updated to 1.0.30, same issue as in 1.0.18 which I used before.

TousstNicolas, 2024-03-07:
Hi, thanks for reporting this issue.

Unless I am mistaken, KiCad does not allow hiding only one pin name, while EasyEDA allows it. In KiCad, the pin name option is defined in the symbol properties and is applied to all pins. 
This is the case for pin 21, which has a hidden pin name on EasyEDA.

For symbols I tested this on, most of them either had no pin name hidden or all of them. Therefore, if one of them is hidden, I apply it on all pins. As a first fix, you can manually edit the symbol property to set the "show pin name" option.

Another solution is instead of hiding all the pin names, to set the pin name to an empty string. It would have the same visual effect, but the pin name information is lost.

Let me know what you think.

markusdd, 2024-03-07:
I think this probably then becomes an issue for many QFN cases because all of them have these middle GND thermal pads.

I would say it is probably not a 'fail safe default' to then hide all pins in the KiCAD output.

closed 2024-03-19 by TousstNicolas

markusdd, 2024-03-19:
Thanks a lot!