TousstNicolas/JLC2KiCad_lib #66: Windows 10, KiCad 7, Cannot Follow Instructions To Plugin Success ,  KicadModTree Not Sure.

ForrestErickson, 2024-03-06:
**Describe the bug**
A clear and concise description of what the bug is.
I have tried to follow the instructions to install on my Windows 10 system and KiCad 7.

I have the folder "JLC2KiCad_lib" in my plugin library
[image]

When in a command windows, I ran the instructions for installing the KicadModTree:
C:\Users\Admin>C:\Program Files\KiCad\7.0\bin\kicad.exe -m pip install KicadModTree
I got errors. So I guessed at the syntax and finaly got some indication that there was an install. See screen dump.
[image]
 

When in the PCB editor I refresh the plugins nothing changes.
[image]

I had expected to see something added to the list just as the Fabrication Tools was added.

TousstNicolas, 2024-03-06:
Jlc2KiCadLib is intended to be a standalone tool and not a KiCad plugin.

Therefor it is expected to not show up as a kicad plugin.

See #58

closed 2024-03-06 by TousstNicolas

ForrestErickson, 2024-03-06:
Thanks for getting back to me. 
I was following instructions from a fork at: https://github.com/Springer-Electronics/JLC2KiCad_lib and I did not realize the scope of your project was different.