TousstNicolas/JLC2KiCad_lib #68: Kicad Symbol is missing geometry

besi, 2024-03-08:
Footoprint: [OmronElectronics-G5LE_1A4DC6](https://jlcpcb.com/partdetail/OmronElectronics-G5LE_1A4DC6/C1564818)

[image]

Command:

    JLC2KiCadLib -dir lib -symbol_lib jlc_symbols -symbol_lib_dir "." -footprint_lib jlc_footprints.pretty' C1564818

Now in Kicad the symbol looks like follows:

[image]

Looking more closely I figured that the geometry has moved quite some distance upward (towards the sky if you will):
[image]

besi, 2024-03-08:
Another example is `C688864` which just hass the labels floating high above

[image]

TousstNicolas, 2024-03-19:
Hi, thank you for reporting this issue. 

This is an issue I am aware of with the arc of circle's geometry. 
Unfortunately I have not had the time to find a fix. 

Any help on this matter would be welcomed.