TousstNicolas/JLC2KiCad_lib #76: Footprint has no courtyard defined

vk2diy, 2024-09-02:
**Describe the bug**
Courtyard areas are not imported/created, leading to errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Import parts to KiCad 8.0.3.
2. Run DRC.
3. Observe for every footprint errors "Footprint has no courtyard defined".

**Expected behavior**
Footprints are approximated with conservative/safe default margins if they are not explicitly defined in the source data.

**Screenshots**
[image]

**Additional context**
Some sample part numbers include `C408533`, `C7428747`, `C2977080`.

vk2diy, 2024-09-02:
Seems there is already some bounding box related code in https://github.com/TousstNicolas/JLC2KiCad_lib/blob/fc17f915c98fb49c650c4b2837356e68b443e4d6/JLC2KiCadLib/footprint/footprint.py#L53 however it is unclear to me how or if this functions. If it did function, a bounding box should be fairly easy to add.

TousstNicolas, 2026-08-16:
Fixed in 1a7bd61, released in v1.3.0. Since EasyEDA does not provide courtyard data (confirmed by checking both the EasyEDA Standard/Pro layer definitions and KiCad's own current EasyEDA importer source, which builds its own courtyard box for the same reason), footprints now get an auto-generated approximate courtyard on F.CrtYd: the bounding box of the F.Cu/B.Cu/F.Paste/B.Paste/F.Mask/B.Mask/F.Fab/Edge.Cuts graphics inflated by 0.25mm and snapped to the 0.01mm KiCad courtyard grid, mirroring KiCad's own EasyEDA import behavior.

closed 2026-08-16 by TousstNicolas