TousstNicolas/JLC2KiCad_lib #78: I get errors while triing to run a command

Obili69, 2024-10-08:
**Describe the bug**
christianobrecht@Christians-MacBook-Pro ~ % JLC2KiCadLib C1337258 /Users/christianobrecht/Documents/KiCad/LIB
2024-10-08 14:29:53,334 - INFO - creating library for component C1337258
2024-10-08 14:29:53,486 - INFO - creating footprint ...
2024-10-08 14:29:53,658 - INFO - Downloading STEP Model ...
2024-10-08 14:29:53,941 - INFO - STEP model created at JLC2KiCad_lib/footprint/packages3d/VQFN-48_L7.0-W7.0-P0.50-BL-EP5.5.step
2024-10-08 14:29:53,942 - INFO - added JLC2KiCad_lib/footprint/packages3d/VQFN-48_L7.0-W7.0-P0.50-BL-EP5.5.step to footprint
2024-10-08 14:29:53,947 - INFO - created 'JLC2KiCad_lib/footprint/VQFN-48_L7.0-W7.0-P0.50-BL-EP5.5.kicad_mod'
2024-10-08 14:29:54,117 - INFO - creating symbol STM32L431CCU6_0 in STM32L431CCU6
2024-10-08 14:29:54,119 - INFO - found component already in STM32L431CCU6, updating STM32L431CCU6
2024-10-08 14:29:54,122 - INFO - creating library for component /Users/christianobrecht/Documents/KiCad/LIB
Traceback (most recent call last):
  File "/Users/christianobrecht/Library/Python/3.11/bin/JLC2KiCadLib", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/christianobrecht/Library/Python/3.11/lib/python/site-packages/JLC2KiCadLib/JLC2KiCadLib.py", line 181, in main
    add_component(component, args)
  File "/Users/christianobrecht/Library/Python/3.11/lib/python/site-packages/JLC2KiCadLib/JLC2KiCadLib.py", line 14, in add_component
    data = json.loads(
           ^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

**Expected behavior**
An output in the library folder

**Additional context**
running on Mac os 15 M1

TousstNicolas, 2025-02-12:
Hi and thanks for reporting this issue.

It looks like the data fetched from the request was not received properly.
I was not able to reproduce the issue on my side and the symbol/footprint are both created as expected.

Does this issue still occurs on your side ?

Obili69, 2025-02-12:
No it works for me now
Thanks

closed 2025-02-13 by TousstNicolas